jedimatt42/tipi #33: tipi.service can't access some resources

ElectricLab, 2017-10-26:
The latest version of tipi.service has tipi.sh running as user: tipi. Is this correct? If so, did you chmod the following devices to be readable/writeable by the tipi user: 
/dev/mem
/dev/gpiomem
Or maybe you addded 'tipi' to kmem and gpio groups?
The service won't start otherwise.

jedimatt42, 2017-10-26:
I added the tipi user to the gpio group as noted in https://github.com/jedimatt42/tipi/blob/master/dsr/python/README.md

the tipi user is only in the groups:
  tipi
  i2c
  input
  gpio

ElectricLab, 2017-10-26:
Ok - I guess I didn't read that README.md. 
When I ran tipi.sh as the tipi user, I got:
/home/tipi/tipi/dsr/python/tipi.sh
wiringPiSetup: Unable to open /dev/mem or /dev/gpiomem: Permission denied.

The permissions on those devices are:
crw-r----- 1 root kmem 1, 1 Oct 25 22:09 /dev/mem
crw-rw---- 1 root gpio 243, 0 Oct 25 22:09 /dev/gpiomem

So it makes sense that adding tipi to gpio works, but not sure about /dev/mem.

jedimatt42, 2017-10-29:
you shouldn't need access to /dev/mem   because we are using /dev/gpiomem.  The error message is offering alternatives.

jedimatt42, 2017-11-18:
I take Pete's testing to show that this is all cleared up.

closed 2017-11-18 by jedimatt42