jedimatt42/tipi #46: Q*BERT filename doesn't translate to TIFILES and TIPI gets ugly.

jedimatt42, 2017-11-29:
Q*BERT doesn't become a good TIFILES named file... do something about it.

ElectricLab, 2017-11-30:
I say we disallow asterisks in filenames, unless there's some compelling reason to support this, like legacy compatibility. Otherwise, people should be used to * being an illegal filename character, or at least a really bad idea.

jedimatt42, 2017-12-24:
so windows doesn't let you create files with the following characters in the name:
/ \ : * ? < > | 

however, TIPI runs on Linux, and has no problem with it if transferred correctly. 

Linux only disallows '/' in filenames. Which by TIPI rules is translated to a '.' and back.

So this should just go away as a problem once rename is handled, and we implement the name translation rules in the webui.

For now, user must ssh in, and use mv to rename files if the name violates Windows restrictions.

```
mv Q*T 'Q*BERT'
mv Q*U 'Q*BERU'
```

jedimatt42, 2017-12-24:
Also, TIPI ignores the name in the TIFILES header. Totally on purpose, per Tursi's specification on TI foreign hosted filesystems.

jedimatt42, 2020-07-23:
This is fixed.. filenames are now unidecoded... which converts them to the closes pronouncable ascii and they get the long-file-name treatment with hashes...

closed 2020-07-23 by jedimatt42