jedimatt42/tipi #66: DSK?_DIR should represent the volume name somehow

jedimatt42, 2017-12-11:
DEADLINE loader expects and validates the volume name of the disk... 

Suggestion is that the last part of the path name should be reported as the volume name when directories are mapped to DSK?_DIR entries.

arcadeshopper, 2017-12-11:
either that or set it in a file in the directory if that's easier.. like .VOLNAME or something

jedimatt42, 2017-12-25:
TIPI.anthing.more.whatever  volume name will be TIPI
DSK0.anything.more.whatever volume name will be TIPI
DSK1.anything... volume name will be basename of directory DSK1_DIR is mapped to:
So, if DSK1_DIR=GAMES.DEADLINE, then DSK1 volume name will be DEADLINE
In this case, DSK1.SAVES. volume name would still be DEADLINE. Meaning volume name will be tied to the drive mapping, not the directory you currently listing.

PR #75 (2017-12-25): set volume name to the directory the drive is mapped to.
fixes #66
commit c05b1fe8 (2017-12-25): set volume name to the directory the drive is mapped to.

fixes #66
closed 2017-12-25 by jedimatt42