jedimatt42/tipi #86: when editing a text file on the ti, changes aren't written to the file completely

arcadeshopper, 2018-01-25:
using ea editor and preditor both i removed 50 lines from a text file and saved it only to find the lines still there on the file.. my specific line changes seemed to save at the top, but the erased lines did not

jedimatt42, 2018-01-26:
That's a cool bug! We could have gone years without noticing that.

arcadeshopper, 2018-01-26:
additional info, when editing a file with ea editor it seems to add a bunch of garbage to the eof as well I just "saved as" auto-bat2 with my shortened file, that worked fine.. then i edited it to switch the order of events and got this after saving:  
[image]

jedimatt42, 2018-02-11:
fixed

closed 2018-02-11 by jedimatt42

arcadeshopper, 2018-02-12:
still getting the same garbage in the file as above