jedimatt42/tipi #91: in xb call load("DSK.THE_VALLEY.FILENAME") gives an error 00

arcadeshopper, 2018-02-22:
program is here: http://www.stuartconner.me.uk/ti/ti.htm#the_valley

works fine for old and run commands but not call load 
¯\_(ツ)_/¯

jedimatt42, 2018-02-23:
call load works fine... but in this game, the file is a v9t9 file, and when extracted from disk image to host filesystem it is being named VALLEY(UPPERBAR)O. The software is trying to load VALLEY/O 

Now, the host filesystem is where the name is... TIPI does not look in headers for names, so you don't have that confusion... once forget what you know about PC99 or V9T9.  If you rename a file that is the name it has.   

The only exception to that is / and \  characters are allowed in TI names, but not allowed on the host filesystem... so to create a file on the host file system that can be found by TI looking for a /, use a '.'  (period)   so in this case, name the file VALLEY.O

I'll keep this open though, because... my goal is that when you upload a V9T9 file through the webui, it fixes this for you automatically.

--- 

I experience another problem running this game after it loads the assembly object file.  "STRING TRUNCATED IN 16030"  which is I think assembly code also embedded in the XB "RUN" program file.

jedimatt42, 2020-07-23:
When the disk extracts, it names the files correctly now.

jedimatt42, 2020-07-23:
The game actually works from TIPI, if the TIPI crubase is >1100 ( emulating floppy ) --- so the call files routine must be required.

closed 2020-07-23 by jedimatt42