jedimatt42/tipi #98: TELNET cursor in 64 column mode is broken.

jedimatt42, 2018-05-29:
Reading the character from the VDP isn't going to work in bitmap mode.

docbain, 2021-07-07:
I installed Raspberry Pi 4 to my TI using the PEB card set at CRU >1000. It connects to my network with IP address, however, when I attempt to CALL TIPI("TIPI.NET.TELNET"), my TI returns to the logo screen. What am I doing wrong. Also, there is a 10 pin connector on the card identical to the connector on the rear of the TIPI PEB card. Can this connector be used to connect the Raspberry PI, or just the rear connector? Thank you.

jedimatt42, 2021-07-07:
> I installed Raspberry Pi 4 to my TI using the PEB card set at CRU >1000. It connects to my network with IP address, however, when I attempt to CALL TIPI("TIPI.NET.TELNET"), my TI returns to the logo [...]

1st: this has nothing to do with the issue of the telnet cursor...  
2nd: this isn't a support channel... you go over here: https://atariage.com/forums/topic/277635-tipi-usage-and-support/  I receive 0% from sales of TIPI - 
3rd: The 2 ports you speak of are far from identical. one is labelled RPI1 and the other is labelled JTAG1   -- this question was recently answered over here: https://atariage.com/forums/topic/277635-tipi-usage-and-support/

Your initial question of what's wrong will take some back and forth. So you can read some troubleshooting tips on the TIPI wiki 

https://github.com/jedimatt42/tipi/wiki

and if that doesn't give you the answers you need, go to https://atariage.com/forums/topic/277635-tipi-usage-and-support/ and your vendor or the community may help you diagnose the problem. 

possible causes of that crash back to title screen could be:

* faulty memory expansion. 
* missing TIPI.NET.TELNEU, as an EA5 program, it is split into 8k chunks, failure to find the second chunk in CALL TIPI, crashes to title screen. 
* possibly bad cabling between TIPI and PI. Although this usually causes a hang or wild crash/lockup.

jedimatt42, 2021-07-07:
I have no intention of fixing the 64 column text mode cursor... so I'm closing this issue.

closed 2021-07-07 by jedimatt42