jedimatt42/tipi #117: saving basic program from web creates program that won't open in basic

arcadeshopper, 2020-01-18:
tested with the create new program icon and also editing an existing program, I pasted in the following:
 100 PRINT "TCP Server Test"
 110 LSOCK$="PI.TCP=*:9974"
 120 OPEN #1:LSOCK$&".BIND"
 130 INPUT #1:H
 140 IF H<>0 THEN 160
 150 GOTO 130
 160 PRINT "CONNECTION!"
 170 PRINT H
 180 OPEN #2:LSOCK$&"."&STR$(H)
 190 PRINT #2:"Hello I am a TI-99/4A"&CHR$(13)&CHR$(10)
 200 FOR I=1 TO 100 :: W=(I*2.2/2.2):: NEXT I
 210 CLOSE #2
 220 GOTO 130
 230 CLOSE #1

then hit save 

ti give i/o error 02 when trying to load..and the program no longer shows up in the web interface as a basic program that can be edited.. (saved as TIPISERVER)
[image]

jedimatt42, 2020-01-18:
It is because you lost the * in line 200 ... I*2.2

But, we should fix the interface to keep your code and show the syntax errors.

arcadeshopper, 2020-01-18:
Two things to fix

arcadeshopper, 2020-01-20:
never-mind this is a feature that if the basic program isn't passing syntax it doesn't save properly it should error not just save a broken program file