jedimatt42/tipi #163: emu_config.sh - Missing.

reed-rosa, 2022-04-17:
On:
https://github.com/jedimatt42/tipi/wiki/emulation-installation
 
It mentions if running with a real PI and and emulator you should:
"Enable emulation mode, by running /home/tipi/tipi/emulation/emu_config.sh and responding to the prompts. On a real Raspberry PI, PDF conversion can be enabled, and you probably don't want tipi_disk mounted via NFS. "
 
I don't have an emu_config.sh.
I have:
root@tipi:/home/tipi/tipi/emulation# ls -al
total 4044
drwxr-xr-x  2 tipi tipi    4096 Dec 27  2020 .
drwxr-xr-x 11 tipi tipi    4096 Nov 21 14:21 ..
-rw-r--r--  1 tipi tipi     144 Dec 27  2020 .gitattributes
-rwxr-xr-x  1 tipi tipi     783 Dec 27  2020 LICENSE.txt
-rw-r--r--  1 tipi tipi     533 Dec 27  2020 README.md
-rwxr-xr-x  1 tipi tipi    1832 Dec 27  2020 SETUP.txt
-rwxr-xr-x  1 tipi tipi  274944 Dec 27  2020 WinNFSd.exe
-rwxr-xr-x  1 tipi tipi 3814920 Dec 27  2020 kernel-qemu-4.19.50-buster
-rwxr-xr-x  1 tipi tipi      56 Dec 27  2020 nfs_tipi_disk.bat
-rwxr-xr-x  1 tipi tipi     352 Dec 27  2020 tipi-qemu.bat
-rwxr-xr-x  1 tipi tipi    8573 Dec 27  2020 versatile-pb-buster.dtb
root@tipi:/home/tipi/tipi/emulation#

jedimatt42, 2022-04-17:
This is my mistake, the docs are ahead of the download-able image. 

They line up if you are on a real raspberry PI that is up to date via TIPICFG, but not from a starting with emulation / QEMU approach. I am working on revamping all that. 

I need to produce a new SD card image for you all to start from.

jedimatt42, 2022-04-17:
the image included with the latest tipi-emulation___.zip has the file in question, and is pre-configured with emulation mode, no nfs, no pdf.

closed 2022-04-17 by jedimatt42