jedimatt42/tipi #173: opening a non-existent file for reading doesn't return an open failure

mcvde, 2024-12-21:
C libti99 dsrlink returns 0 (success) when opening a non-existent file for reading  files. 

This has occurred with 
DSR_READ, DSR_TYPE_DISPLAY | DSR_TYPE_VARIABLE | DSR_TYPE_SEQUENTIAL | DSR_TYPE_INPUT,
DSR_WRITE, DSR_TYPE_INTERNAL | DSR_TYPE_FIXED | DSR_TYPE_RELATIVE | DSR_TYPE_UPDATE, and
DSR_READ, DSR_TYPE_INTERNAL | DSR_TYPE_FIXED | DSR_TYPE_RELATIVE | DSR_TYPE_INPUT.

looking at BasicFile.py, I don't see any test for file open failure. 

The failure occurs on a side car mounted tipi on a real console. The same code works correctly in a non-tipi configured name instance.

mcvde, 2024-12-21:
Sorry the file I was reviewing is NativeFile.py rather than BasicFile.py.

jedimatt42, 2024-12-22:
This is tricky. DSRLNK can accept a return of 0 from the DSR to mean not handled, and most DSRLNKs will move on and try the next card that can handle the device name, or then fail if no card chooses to handle the request. I'll have to peek at the current libti99 DSRLNK. It may be time to rewrite the 'not handled' code to depend solely on drive mapping state instead file found state.

mcvde, 2024-12-22:
I’d forgotten about the unhandled DSR also returning zero. I currently have code in my fopen that confirms the DSR exists, having cached the list as part of the runtime init, before calling the dsrlink() method with DSR_OPEN.

i recall borrowing your code from force command to read the DSRs.

The current dsrlink() will return non-zero for a missing file if the DSR is handled.

jedimatt42, 2024-12-29:
Reviewing the code, I have a couple loopholes where if the file isn't there and it is a read-only operation, it'll pass back to the DSRLNK... I'm working on tightening that up, so it'll return an error to DSRLNK if the drive was mapped ( should have been handled ) It looks like this impacts OPEN and LOAD. 

And I'm adding an option for re-enabling my old behavior. The use case for the old behavior is to allow someone to have DSK1. mapped to a directory that contains the CHARA1 file ( for instance ) and then on their real floppy have the TI-Writer disk or EA disk in, and consequently the CHARA1 from the mapped TIPI drive will load instead of the one in the floppy drive. This allows overriding the file ( in this case the 40 column font ) - I only know of one person who liked that feature, and they don't do anything productive with their 4a anyway - I believe this was behavior borrowed from the Myarc HFDC harddrive controller.

mcvde, 2024-12-30:
That sounds like a great solution. I look forward to it.

jedimatt42, 2025-01-05:
This update released as v4.4 is the fix: https://github.com/jedimatt42/tipi/commit/6723cc4ad690ab07a1ca8dbdb4490df8a3dcf45a

mcvde, 2025-04-05:
It's taken me three months to get to checking out the fix, between side projects and getting DSR ROM updates. The SW load I started with was 4.4 but it didn't seem to fix this, odd. But the TIPI configuration screen popped up with an available update to 4.6 and I let it load it. After that non-existent files reported appropriately.

The reason this was a big deal to me was that many commands in my OS are typed at the command line. Any misspellings of either the command or file arguments meant a system crash. Very happy to have this fix. Thank you!