jedimatt42/tipi #175: cant upload backup

arcadeshopper, 2025-02-13:
OSError
OSError: [Errno 28] No space left on device

Traceback (most recent call last)
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 1498, in __call__
return self.wsgi_app(environ, start_response)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask_socketio/__init__.py", line 43, in __call__
return super(_SocketIOMiddleware, self).__call__(environ,
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/engineio/middleware.py", line 74, in __call__
return self.wsgi_app(environ, start_response)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 1476, in wsgi_app
response = self.handle_exception(e)
           ^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
response = self.full_dispatch_request()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
rv = self.handle_user_exception(e)
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
rv = self.dispatch_request()
     ^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/./route.py", line 447, in backupul
tipi_backup.upload(request.files.getlist("upload_file"))
                   ^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/utils.py", line 107, in __get__
value = self.fget(obj)  # type: ignore
        ^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 494, in files
self._load_form_data()
^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/flask/wrappers.py", line 115, in _load_form_data
super()._load_form_data()
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 268, in _load_form_data
data = parser.parse(
       
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/formparser.py", line 242, in parse
return parse_func(stream, mimetype, content_length, options)
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/formparser.py", line 267, in _parse_multipart
form, files = parser.parse(stream, boundary, content_length)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/tipi/tipi/htdocs/ENV/lib/python3.11/site-packages/werkzeug/formparser.py", line 384, in parse
_write(event.data)
^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/tempfile.py", line 966, in write
self._check(file)
^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/tempfile.py", line 834, in _check
self.rollover()
^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/tempfile.py", line 846, in rollover
newfile.write(file.getvalue())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 28] No space left on device
The debugger caught an exception in your WSGI application. You can now look at the traceback which led to the error.
To switch between the interactive traceback and the plaintext one, you can click on the "Traceback" headline. From the text traceback you can also create a paste of it.

tipi@tipipeb:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            318M     0  318M   0% /dev
tmpfs            91M  4.4M   87M   5% /run
/dev/mmcblk0p2   29G  2.9G   25G  11% /
tmpfs           454M     0  454M   0% /dev/shm
tmpfs           5.0M   12K  5.0M   1% /run/lock
tmpfs           100M  100M     0 100% /tmp
tmpfs           100M  160K  100M   1% /var/log
/dev/mmcblk0p1  510M   64M  447M  13% /boot/firmware
tmpfs            91M     0   91M   0% /run/user/1000

arcadeshopper, 2025-02-13:
i am able to restore files via samba..but ..

jedimatt42, 2025-02-15:
the backup archive needs to end up in the /home/tipi folder on the PI, and then it should be selectable for restore. SFTP or SCP it there. 

Looks like the web framework is uploading to the /tmp folder on it's way to the real location, which is out of space.

jedimatt42, 2025-02-15:
fixed by https://github.com/jedimatt42/tipi/commit/f47f9e00da1110ed34ebf7a71a62c06ac7eda038