jedimatt42/tipi #184: uhoh

arcadeshopper, 2025-08-05:
ipi@tipi:~/tipi/setup $ ./upgrade.sh --upgrade
Current Version: 4.2
Latest Version: 4.6
Password: 
remote: Enumerating objects: 2611, done.
remote: Counting objects: 100% (248/248), done.
remote: Compressing objects: 100% (200/200), done.
remote: Total 2611 (delta 165), reused 121 (delta 48), pack-reused 2363 (from 2)
Receiving objects: 100% (2611/2611), 2.01 MiB | 3.35 MiB/s, done.
Resolving deltas: 100% (620/620), completed with 7 local objects.
error: inflate: data stream error (incorrect header check)
error: unable to unpack 178d9eba8372da44af371157f464bdc10336b5f5 header
fatal: loose object 178d9eba8372da44af371157f464bdc10336b5f5 (stored in .git/objects/17/8d9eba8372da44af371157f464bdc10336b5f5) is corrupt
error: https://github.com/jedimatt42/tipi.git did not send all necessary objects

arcadeshopper, 2025-08-05:
this is a newly burned sdcard,  first update  tested rw with storing a program in basic

closed 2025-08-05 by arcadeshopper

webdeck, 2025-08-05:
Possibly related to https://github.com/piwheels/packages/issues/530 ?